zhgzhg/LoRaPacketForwarder #1: Unable to execute the program

Pr-Jay, 2019-06-10:
Hello,
So I found out that the ./LoRaPktFwrd should be the command to execute the program, not   ./LoRaPktFwd. But even then it outputs "Unable to open SPI device: No such file or directory". I do have SPI enabled on my Raspberry Pi 3, can you help me find out why it's not working?
Here is the full output:

PlatformInfo:
------------

ID (EUI-64): b8:27:eb:ff:ff:bc:f7:3a

(WiringPI) Pins:
  nss_cs=4
  dio0=5
  dio1=6

LoRa SX127x Chip:
  Freq=434.000000 MHz
  BW=125.000000 KHz
  SF=7
  CR=4/5
  SyncWord=0x34
  PreambleLength=8

Meta Information:
  Latitude=0.000000
  Longtitude=0.000000
  Altitude=10 meters
  Name/Definition: 1chan_uplink_pkt_fwd
  E-mail: contact@email.com
  Description: OPiPC LoRa 1-Ch GW

Unable to open SPI device: No such file or directory

zhgzhg, 2019-06-10:
Hello @Pr-Jay ,

Does the following command show your SPI device:
`ls -la /dev/spi*`

Also can you please provide the output of those commands as well:
`dmesg | grep spi`
`lsmod`

zhgzhg, 2019-06-10:
Also please try executing LoRaPktFwd with sudo to check whether it will make any difference.

Pr-Jay, 2019-06-10:
@zhgzhg  Thanks for replying,
Neither `./LoRaPktFwd` nor `sudo ./LoRaPktFwd` executes the program. But both `./LoRaPktFwrd` and `sudo ./LoRaPktFwrd` do.
[image]
[image]
Here is the result from `ls -la /dev/spi*`, `dmesg| grep spi` and `lsmod`
[image]
Also, here is the screenshot of the directory, in case something is wrong.
[image]
Thanks!

zhgzhg, 2019-06-10:
Hmm. your environment seems to be fine.
With `sudo ./LoRaPktFwrd` do you see the same error message?

Pr-Jay, 2019-06-10:
@zhgzhg Yes, it has the same problem
[image]

zhgzhg, 2019-06-10:
Weird indeed. Currently I have no clue. Looks like for some reason although the SPI device is available there's no access to it. To make sure that it's at least operable I'd suggest testing it in the following way:

[https://www.raspberrypi.org/documentation/hardware/raspberrypi/spi/README.md#troubleshooting](https://www.raspberrypi.org/documentation/hardware/raspberrypi/spi/README.md#troubleshooting)

Please let me know whether you observe different results.

Pr-Jay, 2019-06-10:
@zhgzhg I think it turned out ok
[image]
Could it be that OrangePi and Pi3 WiringPi libaries are slightly different for SPI be the culprit?
[image]

zhgzhg, 2019-06-10:
@Pr-Jay , yes they should differ, and its possible that might be the case. But I am also assuming you are using the proper wiringPi version - the one provided by Raspbian, and not the one bundled in this project, right?

zhgzhg, 2019-06-10:
Another idea - I have noticed that LoRaLib attempts to transfer SPI data with higher than the usual clock speed, so reducing it might have a positive effect. @Pr-Jay , could please execute the following command into the project's directory:

`sed -i 's/SPISettings(2000000/SPISettings(500000/' ./LoRaLib/src/Module.cpp`

and then recompile the project with `make` and try again?

Pr-Jay, 2019-06-11:
@zhgzhg I'm not sure if I'm doing this correctly, it didn't return anything.
[image]
I don't think I'm using wiringPi that came with the project, as I didn't do anything with the wiringPi folders.

zhgzhg, 2019-06-11:
Thanks @Pr-Jay. The sed command executed just fine. I am starting to think that wiringPi is not even addressing the correct SPI path.

Could you please install the _strace_ program:

`sudo apt-get install strace`

and then run LoRaPktFwrd with it redirecting the output to text file in the following way:

`sudo strace -o stacktrace.txt ./LoRaPktFwrd`

and attach the trace log from the newly created _stacktrace.txt_ file here?

Pr-Jay, 2019-06-11:
@zhgzhg Hi,
Here is the content of _stacktrace.txt_:

execve("./LoRaPktFwrd", ["./LoRaPktFwrd"], [/* 15 vars */]) = 0
brk(NULL)                               = 0x7b9000
uname({sysname="Linux", nodename="raspberrypi", ...}) = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
mmap2(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x76f74000
access("/etc/ld.so.preload", R_OK)      = 0
open("/etc/ld.so.preload", O_RDONLY|O_CLOEXEC) = 3
fstat64(3, {st_mode=S_IFREG|0644, st_size=42, ...}) = 0
mmap2(NULL, 42, PROT_READ|PROT_WRITE, MAP_PRIVATE, 3, 0) = 0x76f73000
close(3)                                = 0
open("/usr/lib/arm-linux-gnueabihf/libarmmem.so", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\210\5\0\0004\0\0\0"..., 512) = 512
lseek(3, 20868, SEEK_SET)               = 20868
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1000) = 1000
lseek(3, 20540, SEEK_SET)               = 20540
read(3, "A,\0\0\0aeabi\0\1\"\0\0\0\0056\0\6\6\10\1\t\1\n\3\f\1\22\4\24"..., 45) = 45
fstat64(3, {st_mode=S_IFREG|0644, st_size=21868, ...}) = 0
mmap2(NULL, 86080, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x76f30000
mprotect(0x76f35000, 61440, PROT_NONE)  = 0
mmap2(0x76f44000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x4000) = 0x76f44000
close(3)                                = 0
munmap(0x76f73000, 42)                  = 0
open("/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
fstat64(3, {st_mode=S_IFREG|0644, st_size=85828, ...}) = 0
mmap2(NULL, 85828, PROT_READ, MAP_PRIVATE, 3, 0) = 0x76f1b000
close(3)                                = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
open("/usr/lib/libwiringPi.so", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0x,\0\0004\0\0\0"..., 512) = 512
lseek(3, 69628, SEEK_SET)               = 69628
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1080) = 1080
lseek(3, 51803, SEEK_SET)               = 51803
read(3, "A.\0\0\0aeabi\0\1$\0\0\0\0056\0\6\6\10\1\t\1\n\2\22\4\24\1\25"..., 47) = 47
fstat64(3, {st_mode=S_IFREG|0755, st_size=70708, ...}) = 0
mmap2(NULL, 121320, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x76efd000
mprotect(0x76f09000, 65536, PROT_NONE)  = 0
mmap2(0x76f19000, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0xc000) = 0x76f19000
mmap2(0x76f1a000, 2536, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x76f1a000
close(3)                                = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
open("/usr/lib/arm-linux-gnueabihf/libstdc++.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\3\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0000\332\6\0004\0\0\0"..., 512) = 512
lseek(3, 1268340, SEEK_SET)             = 1268340
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1280) = 1280
lseek(3, 1267740, SEEK_SET)             = 1267740
read(3, "A.\0\0\0aeabi\0\1$\0\0\0\0056\0\6\6\10\1\t\1\n\2\22\4\23\1\24"..., 47) = 47
fstat64(3, {st_mode=S_IFREG|0644, st_size=1269620, ...}) = 0
mmap2(NULL, 1341504, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x76db5000
mprotect(0x76ee4000, 65536, PROT_NONE)  = 0
mmap2(0x76ef4000, 28672, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x12f000) = 0x76ef4000
mmap2(0x76efb000, 6208, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x76efb000
close(3)                                = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
open("/lib/arm-linux-gnueabihf/libm.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0`=\0\0004\0\0\0"..., 512) = 512
lseek(3, 451032, SEEK_SET)              = 451032
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 1120) = 1120
lseek(3, 450676, SEEK_SET)              = 450676
read(3, "A0\0\0\0aeabi\0\1&\0\0\0\0056\0\6\6\10\1\t\1\n\2\22\4\23\1\24"..., 49) = 49
fstat64(3, {st_mode=S_IFREG|0644, st_size=452152, ...}) = 0
mmap2(NULL, 516220, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x76d36000
mprotect(0x76da3000, 65536, PROT_NONE)  = 0
mmap2(0x76db3000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x6d000) = 0x76db3000
close(3)                                = 0
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
open("/lib/arm-linux-gnueabihf/libgcc_s.so.1", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\200\321\0\0004\0\0\0"..., 512) = 512
[... 121 lines cut ...]
write(1, "  nss_cs=4\n", 11)            = 11
write(1, "  dio0=5\n", 9)               = 9
write(1, "  dio1=6\n\n", 10)            = 10
write(1, "LoRa SX127x Chip:\n", 18)     = 18
write(1, "  Freq=434.000000 MHz\n", 22) = 22
write(1, "  BW=125.000000 KHz\n", 20)   = 20
write(1, "  SF=7\n", 7)                 = 7
write(1, "  CR=4/5\n", 9)               = 9
write(1, "  SyncWord=0x34\n", 16)       = 16
write(1, "  PreambleLength=8\n\n", 20)  = 20
write(1, "Meta Information:\n", 18)     = 18
write(1, "  Latitude=0.000000\n", 20)   = 20
write(1, "  Longtitude=0.000000\n", 22) = 22
write(1, "  Altitude=10 meters\n", 21)  = 21
write(1, "  Name/Definition: 1chan_uplink_"..., 40) = 40
write(1, "  E-mail: contact@email.com\n", 28) = 28
write(1, "  Description: OPiPC LoRa 1-Ch G"..., 35) = 35
open("/dev/spidev0.0", O_RDWR)          = 5
ioctl(5, SPI_IOC_WR_MODE, 0x7e8313b0)   = 0
ioctl(5, SPI_IOC_WR_BITS_PER_WORD, 0x76f0868a) = 0
ioctl(5, SPI_IOC_WR_MAX_SPEED_HZ, 0x7e8313b4) = 0
open("/dev/spidev0.500000", O_RDWR)     = -1 ENOENT (No such file or directory)
write(2, "Unable to open SPI device: No su"..., 53) = 53
exit_group(1)                           = ?
+++ exited with 1 +++

zhgzhg, 2019-06-11:
I think I have found the issue (in LoRaLib port for linux), and it should be fixed now.
@Pr-Jay , you can try pulling the latest changes, or cloning and compiling it from scratch.

Pr-Jay, 2019-06-12:
Thanks, @zhgzhg , there is another problem though. It's now saying  "Unable to open SPI device: Too many open files" 
[image]

zhgzhg, 2019-06-12:
Thanks for the feedback, @Pr-Jay . Hopefully this should be fixed now. Currently I don't have access to real hardware setup to validate properly, but in the following days I'll have that opportunity.

Pr-Jay, 2019-06-12:
@zhgzhg Yes, the program is running now. I haven't test it properly with TTN yet,  I will do that as soon as I'm able. Btw, can I print out the incoming loraDataPacket on to the terminal?
Much thanks for your commitment to resolve this issue.

zhgzhg, 2019-06-12:
You are welcome, @Pr-Jay , and thanks for contributing! Regarding your question about terminal logging - yes, it does that by default - when a packet is received it will log its payload content in the terminal along with signal stats like RSSI and SNR.
If you have a spare SX1278 module and ESP8266/ESP32/Arduino you can flash the Transmit project to generate some LoRa packets.

Pr-Jay, 2019-06-13:
@zhgzhg I believe the LoRa functionality is running fine now.
[image]
A tiny note is that the default Transmit.ino wasn't written to send  at SF 7, and CR 5 like the gateway.
[image]
Also, I have tried to registered the gateway on TheThingsNetwork, but they don't have support for 434MHz. How did you overcome this?

zhgzhg, 2019-06-13:
Great to see that! Obviously I need to correct the comments/config in the Transmit sketch. Regarding 433 MHz band - the standardization process is not complete yet, hence TTN does not provide that option. However this is just a meta-data, thus you are able pick another option (although it is not a great practice, but for testing purposes I suppose it's fine as long as the country you are located in permits the frequency).

Pr-Jay, 2019-06-13:
@zhgzhg Thanks so much for the support! I just uploaded a few first packages to TTN. I would say that this is case closed.

commit 652180f7 (2019-06-11): Upgrade to newer LoRaLib in order to fix #1
closed 2019-06-14 by Pr-Jay