zhgzhg/LoRaPacketForwarder #2: How can I configure the program to receive byte arrays

Pr-Jay, 2019-06-16:
Hello again,
I have been trying to configure the program to accept byte arrays, but I think I have taken on more than I can handle. These are the places that I have edited:
[image]
Here is the file it self
[main.txt](https://github.com/zhgzhg/LoRaPacketForwarder/files/3294192/main.txt)
Hope you could shed some light on this one.

zhgzhg, 2019-06-16:
Hi @Pr-Jay , you can try [this slightly modified version](https://github.com/zhgzhg/LoRaPacketForwarder/files/3294280/main_bytearr.txt). Please keep in mind that receiving byte arrays requires knowing in advance the size of the data you are going to receive, and is also limited by LoRaLib up to 255 bytes.

Pr-Jay, 2019-06-17:
@zhgzhg It's still showing my the nonsense it was receiving before. This is what I get when I tried to receive byte arrays from Transmit.ino with the default main.cpp
[image]

zhgzhg, 2019-06-17:
I am not sure that I've understood the question - from your screenshot I can see you are sending the following hexadecimal bytes: 01 23 45 56 78 AB CD EF, which when printed in ASCII characters format correspond to #EVx«Íï .

Pr-Jay, 2019-06-17:
@zhgzhg Ah, my bad. I wasn't able to interpreted the incomings. That coupled with me trying to send data through another node using arduino-lmic wtih ABP activation made the received data looks nothing like other radio LoRa packets carrying the same thing.
[image]
Thanks for clearing that up. 
Unless I'm doing something else wrong, I don't suppose I could forward data through a registered application on TTN.

closed 2019-06-17 by Pr-Jay

zhgzhg, 2019-06-17:
Yeah, currently this project covers only LoRa and uplink data in particular while no attempts have been made in LoRaWan direction yet.