zhgzhg/LoRaPacketForwarder #3: LoRa chip setup failed, code 65524

augisbud, 2019-08-01:
Hello I get this error when trying to execute the program.
```
LoRaPktFwrd: smtUdpPacketForwarder/rapidjson/document.h:1053: rapidjson::GenericValue<Encoding, Allocator>& rapidjson::GenericValue<Encoding, Allocator>::operator[](const rapidjson::GenericValue<Encoding, SourceAllocator>&) [with SourceAllocator = rapidjson::MemoryPoolAllocator<>; Encoding = rapidjson::UTF8<>; Allocator = rapidjson::MemoryPoolAllocator<>]: Assertion `false' failed.
Aborted
```
I also used to get this error

`LoRa chip setup failed, code 65524`

I'm using a Dragino Lora/Gps HAT

my config.json
```
{
  "spi_speed_hz": 2000000,
  "spi_channel": 0,

  "pin_nss_cs": 6,
  "pin_dio0": 7,
  "pin_dio0": 0,
  "pin_rest": 0,

  "spreading_factor": 12,
  "carrier_frequency_mhz": 868.1,
  "bandwidth_khz": 125.0,
  "coding_rate": 5,
  "sync_word": 52,
  "preamble_length": 8,

  "latitude": 55.07140615,
  "longtitude": 24.27896652,
  "altitude_meters": 20,

  "platform_definition": "1chan_uplink_pkt_fwd",
  "platform_email": "contact@email.com",
  "platform_description": "OPiPC LoRa 1-Ch GW",

  "servers": [
    {
      "address": "52.169.76.203",
      "port": 1700,
      "enabled": true
    },
    {
      "address": "router.us.thethings.network",
      "port": 1700,
      "enabled": false
    }
  ]
}
```

zhgzhg, 2019-08-01:
Hello, from your configuration I can see that "pin_dio0" is specified twice instead of having "pin_dio0" and "pin_dio1".  The reset/restart pin ("pin_rest") should be also set to a negative value if it's not used or to something different than the values specified for the "pin" fields.

Currently I am not very confident that this project is suitable/would work with your Dragino hat.
The current support is only for the SX1278 chip (although I plan to extend it in the future) while Draginos on 868.1 MHz typically use SX1276.

augisbud, 2019-08-01:
> Hello, from your configuration I can see that "pin_dio0" is specified twice instead of having "pin_dio0" and "pin_dio1". The reset/restart pin ("pin_rest") should be also set to a negative value if [...]

yup, fixed the config, now i get the 
`LoRa chip setup failed, code 65524`
error.
Any ideas, could I use a RFM95?

augisbud, 2019-08-01:
the Dragino Hat uses SX1276, it works on another code, check out my profile forked
https://github.com/augisbud/single_chan_pkt_fwd

zhgzhg, 2019-08-01:
Thanks for the link.
On your question - the short answer is not yet. On theory the LoRa library this project is based on should support it, but there's more work to be done...

augisbud, 2019-08-01:
> Thanks for the link.
> On your question - the short answer is not yet. On theory the LoRa library this project is based on should support it, but there's more work to be done...

can you tell me the difference between SX1272/76/78 and where could I acquire SX1278

zhgzhg, 2019-08-01:
Section "1.2 Product Versions" / page 10 in Semtech's datasheet answers that - [`https://www.semtech.com/uploads/documents/DS_SX1276-7-8-9_W_APP_V6.pdf`](https://www.semtech.com/uploads/documents/DS_SX1276-7-8-9_W_APP_V6.pdf)

[This discussion](https://www.thethingsnetwork.org/forum/t/sx1272-versus-sx1276-receiver-sensitivity/5927/2) also covers part of that information.

However it really comes to the concrete module you are buying and the particular frequency it is optimised for. (The manufacturer accomplishes that by tuning the values of the passives controlling the transmitting frequency).

You can search in eBay for modules like "SX1278 RA-02" and similar ones. For e.g.:

[https://www.ebay.com/itm/SX1278-LoRa-Module-433M-10KM-Ra-02-Wireless-Spread-Socket/392167020016?hash=item5b4ef9d5f0:g:ymAAAOSwMg5b3~7E](https://www.ebay.com/itm/SX1278-LoRa-Module-433M-10KM-Ra-02-Wireless-Spread-Socket/392167020016?hash=item5b4ef9d5f0:g:ymAAAOSwMg5b3~7E)

[https://www.ebay.com/itm/LoRa-SX1278-Long-Range-RF-Wireless-Power-Mental-Module-SX1276-5Km-For-Arduino/323355981969?hash=item4b4984b891:g:ykkAAOSw39NbV-oE](https://www.ebay.com/itm/LoRa-SX1278-Long-Range-RF-Wireless-Power-Mental-Module-SX1276-5Km-For-Arduino/323355981969?hash=item4b4984b891:g:ykkAAOSw39NbV-oE)

augisbud, 2019-08-02:
> Section "1.2 Product Versions" / page 10 in Semtech's datasheet answers that - https://www.semtech.com/uploads/documents/DS_SX1276-7-8-9_W_APP_V6.pdf
>
> [This [...]

you live in europe but seem to use the module meant for the asian market?

augisbud, 2019-08-02:
also one more thing, does you code support SF12BW500 or atleast SF12BW250

zhgzhg, 2019-08-02:
It also depends on your country's local regulations. In some EU countries 433 MHz is legal too. Regarding the SF12BW500 and SF12BW250 - yes, it is supported.

zhgzhg, 2019-08-04:
@augisbud , I've made the configuration to require "ic_model" field, specifyng the exact LoRa chip model used. List of the valid options can be found in the README.md of the project. You can also refer to the config.json.template file for an example.

The new option along with the existing ones should provide wider support for the SX and RFM 
 ICs family. Just make sure that you've properly configured "pin_dio1" for Dragino, otherwise the initialisation will certainly fail.

augisbud, 2019-08-05:
> @augisbud , I've made the configuration to require "ic_model" field, specifyng the exact LoRa chip model used. List of the valid options can be found in the README.md of the project. You can also [...]

the dragino shield doesn't have a dio1 wired, so i should do it manually to one of the gpios?

augisbud, 2019-08-05:
soldered dio1, i get no errors however from the transmit code I don't get a response.
```
Target LoRa Chip Model: SX1276

SPI Settings:
  SPI channel=0
  SPI clock speed=2000000 Hz

(WiringPI) Pins:
  nss_cs=6
  dio0=7
  dio1=5
  rest=-1

LoRa SX1276 Chip:
  Freq=868.099976 MHz
  BW=125.000000 KHz
  SF=12
  CR=4/5
  SyncWord=0x34
  PreambleLength=8

Meta Information:
  Latitude=0.000000
  Longtitude=0.000000
  Altitude=20 meters
  Name/Definition: RPI
  E-mail: info@ginetas.lt
  Description: PktFWD

LoRa chip setup success!
Sending stat update to server(s)... done
```
in the arduino
`int state = lora.begin(868.1, 125.0, 12, 5, 0x34);`

zhgzhg, 2019-08-05:
Soldering dio1 was required in your case.

Regarding the Transmit sketch in Arduino - around line 32 there should be the following code:

`SX1278 lora = new LoRa(NSS, DIO0, DIO1);`

Can you try changing it to:

`SX1276 lora = new LoRa(NSS, DIO0, DIO1);`

augisbud, 2019-08-05:
> Soldering dio1 was required in your case.
> 
> Regarding the Transmit sketch in Arduino - around line 32 there should be the following code:
> 
> `SX1278 lora = new LoRa(NSS, DIO0, DIO1);`
> 
> Can you try changing it to:
> 
> `SX1276 lora = new LoRa(NSS, DIO0, DIO1);`

yes I have done this and forgot to mention

zhgzhg, 2019-08-05:
That's strange. I guess it's needless to mention double checking NSS, DIO0, DIO1 pin definitions and wiring.

You can also enable debug mode in Transmit.ino by adding at the top:

`#define RADIOLIB_DEBUG`

The same debug mode can be enabled in the packet forwarder by compiling it with:
`make debug`

It should show some extra information in the console.

augisbud, 2019-08-05:
enabled debugging and now everything works, thanks

zhgzhg, 2019-08-05:
Could you try configuring both sides with lower SF and BW settings, while using the non-debugging version of the projects?

augisbud, 2019-08-05:
yes

zhgzhg, 2019-08-05:
Does the LoRa communication work?

augisbud, 2019-08-06:
yes

zhgzhg, 2019-08-07:
Looks like the problem is resolved. I am closing the ticket.

closed 2019-08-07 by zhgzhg