zhgzhg/LoRaPacketForwarder #6: Use with Lora Pi hat

superjasiek, 2021-03-08:
Hi i'm trying tu use this library with waveshare lora hat (https://www.waveshare.com/wiki/SX1268_433M_LoRa_HAT) based on sx1268 chip but unfortunately Pi does not recognize it - i've changed ic_model:  to SX1268 but it does not work. This chip is based on UART interface - E22-400T22S consisting rx, tx, aux, m0 and m1 pins. Would you be so kind and give me some tips how to get it working?
Take care!

Di-Ny, 2021-03-08:
Hi, maybe the maintener will  correct me, but your module is an Ebyte module that is made to communicate over the LoRa protocol with other Ebytes nodes. It does not integrates the LoRa**WAN** stack, and last time I checked it was not possible to make use of it for LoRaWAN. 
This repo is made to interface an Orange Pi with a bare SX1276 (or an other) implementing LoRaWAN, and talking over SPI.
In other words, it will not be possibile to use this module with TTN or Chripstack. The kind of product you're looking for is this [one](https://fr.aliexpress.com/item/32856223660.html).

superjasiek, 2021-03-08:
Yes it is but in general used library (RadioLib) includes sx1268 chip and it's also compitable w sx1278  - sth like [this](https://www.hackster.io/ChrisSamuelson/lora-raspberry-pi-single-channel-gateway-cheap-d57d36) So i rhout that maybe it is possible to compilie forwarder for this kind of device. Thank's for your interest!

zhgzhg, 2021-03-10:
Hi @superjasiek , unfortunately the Ebyte modules won't work with RadioLib, and hence this project. As @Di-Ny correctly stated they are narrowly specialized, easing particular communication scenarios.

As far as I understand they don't offer direct communication over SPI with the SX126x chip, but instead provide a UART proxy with simpler config/communication protocol. Also RadioLib currently supports only direct communication with the SX and RFM chips over the SPI. And I couldn't see in E22's documentation whether is possible to set low-level transmit/receive parameters like spreading factor, and coding rate, which are mandatory to achieve LoRaWAN interoperability.

If you need a LoRaWAN packet forwarder you can either go with a simpler hat, based on the pure LoRa chip, or even if you don't mind some soldering you can use something like [this](https://www.aliexpress.com/item/33057410509), [that](https://www.aliexpress.com/item/4000076778363.html), or similar...

superjasiek, 2021-03-10:
Thank you for u r replay. There is some kind of configuartion tool on Ebyte site that configures device over UART [here](https://www.ebyte.com/en/data-download.html?page=1&id=37&cid=31#load).  It has quite a lot of paramteres but unfortunately i don't see spreding factor or coding rate. So according to you r comment i asume that in general this kind of hats ar useless for forwarding purpose and in general should be used in p2p communictaion.
Thank's one more time!

closed 2021-03-13 by zhgzhg