zhgzhg/LoRaPacketForwarder #7: Make stuck on RaspberryPi Zero W

moorgrove, 2021-07-15:
I´m trying to install it on a RaspberryPi Zero W but Make get stuck on:

g++ -o LoRaPktFwrd RadioLib/src/linux-workarounds/SPI/SPI.cpp RadioLib/src/linux-workarounds/dummy.cpp RadioLib/src/protocols/PhysicalLayer/PhysicalLayer.cpp RadioLib/src/modules/RF69/RF69.cpp RadioLib/src/modules/RFM9x/RFM96.cpp RadioLib/src/modules/RFM9x/RFM95.cpp RadioLib/src/modules/RFM9x/RFM97.cpp RadioLib/src/modules/SX127x/SX1276.cpp RadioLib/src/modules/SX127x/SX1273.cpp RadioLib/src/modules/SX127x/SX1278.cpp RadioLib/src/modules/SX127x/SX1279.cpp RadioLib/src/modules/SX127x/SX1277.cpp RadioLib/src/modules/SX127x/SX127x.cpp RadioLib/src/modules/SX127x/SX1272.cpp RadioLib/src/modules/SX126x/SX1262.cpp RadioLib/src/modules/SX126x/SX126x.cpp RadioLib/src/modules/SX126x/SX1268.cpp RadioLib/src/modules/SX126x/SX1261.cpp RadioLib/src/ISerial.cpp RadioLib/src/Module.cpp smtUdpPacketForwarder/base64/base64.c smtUdpPacketForwarder/gpsTimestampUtils/GpsTimestampUtils.cpp smtUdpPacketForwarder/Radio.cpp smtUdpPacketForwarder/TimeUtils.cpp smtUdpPacketForwarder/UdpUtils.cpp smtUdpPacketForwarder/ConfigFileParser.cpp main.cpp -std=c++14 -Wall -DLINUX -DARDUINO=999 -IRadioLib/src/ -IRadioLib/src/modules/ -IRadioLib/src/protocols/ -IRadioLib/src/modules/RF69/ -IRadioLib/src/modules/RFM9x/ -IRadioLib/src/modules/SX127x/ -IRadioLib/src/modules/SX126x/ -IRadioLib/src/linux-workarounds/ -IRadioLib/src/linux-workarounds/SPI/ -O3 -lwiringPi -lm -lpthread -lrt -lcrypt
Putting child 0x1aa7f90 (all) PID 1168 on the chain.
**Live child 0x1aa7f90 (all) PID 1168**

moorgrove, 2021-07-15:
Was not patient enough :)

closed 2021-07-15 by moorgrove

zhgzhg, 2021-07-16:
It usually takes some time even on more powerful models.