zhgzhg/LoRaPacketForwarder #8: RFM96 error code

dogefreak, 2021-08-20:
I'm currently using this code on my RbPI together with an RFM96, configured the pins correctly, updated WiringPi ect...
Getting error code "65520" (LoRa chip setup failed, code 65520). Any suggestions on what this could be?

zhgzhg, 2021-08-21:
Error 65520 indicates unsuccessful SPI write operation. Please check the SPI MISO and MOSI pins. Connecting them in reverse can cause such an issue.

zhgzhg, 2021-09-01:
@dogefreak did you manage to resolve the problem?

dogefreak, 2021-09-01:
@zhgzhg I double-checked everything, including the pins. Everything seems to be right. It'd be useful to know what the error codes actually mean. For instance, I've had 65520 but now I have 65524.

zhgzhg, 2021-09-01:
I'll certainly improve that part.

Error 65524 indicates bad frequency - perhaps your config's _carrier_frequency_mhz_ value has it set outside the supported for RFM96 range,

dogefreak, 2021-09-01:
Yepp I got it. Some dodgy vendor sold me the wrong RFM module. I suggest, if you have time to implement readable error codes you definitely should. Thanks for the great project & support, closing the issue now.

closed 2021-09-01 by dogefreak