zhgzhg/LoRaPacketForwarder #14: SX 1262

batsuurib, 2024-01-04:
How are you?

I used heltec HT-RA62 /SX 1262/. SX1276 was normal.
The following error occurred.

```
LoRa chip setup failed, code (64829) Failed to execute an SPI command. Make sure the module is not using TCXO while still having its XTAL connected.
```

What did I do wrong? Help me.

Is it correct to assume that the SX1262 will receive data from further away?

zhgzhg, 2024-01-04:
Hi @batsuurib ,
Thanks for reporting this!

With the latest RadioLib upgrades the initialisation of the SX126x series got changed a bit, so I had to readjust it in my latest commit.

Could you please give it a try?

zhgzhg, 2024-01-28:
I am assuming the issue is gone.
Feel free to reopen if that's not the case.

closed 2024-01-28 by zhgzhg