zhgzhg/LoRaPacketForwarder #15: Chirpstack v4

batsuurib, 2024-03-18:
Hello

Can LoRaPacketForwarder be used by chirpstack v4?
The problem is that chirpstack v4 requires Gateway ID DevEUI 64.

zhgzhg, 2024-03-19:
I have not tested it with v4, but it should be compatible.

The forwarder's gateway ID is printed in the terminal when it's started and is based on the MAC address of the network adapter used by it for the data transmission.

It should look something like this in the logs:

```
(Tue Mar 19 07:15:25 2024) Started /usr/bin/LoRaPktFwrd...
PlatformInfo:
------------
LoRa Packet Forwarder Version: next-5dd7e0d
ID (EUI-64): 0d:ab:99:ff:ff:cd:32:d6
Target LoRa Chip Model: SX1278
SPI Settings:
  SPI port=0
  SPI channel=1
  SPI clock speed=2000000 Hz
...
...
...
LoRa chip setup succeeded! Waiting for data...
```
So in this example `0d:ab:99:ff:ff:cd:32:d6` is the Gateway ID DevEUI 64.

On some Linux distros the MAC address randomization is turned on by default which can cause changing of the DevEUI 64 after every device reboot. In that case turning off the randomization will fixate the id.

batsuurib, 2024-03-19:
Thank you,

I didn't know that. Try it.

closed 2024-04-08 by zhgzhg