zhgzhg/LoRaPacketForwarder #17: 1ch GW for Chirpstack NEtwork Server on RPI4 + Waveshare SX1262 LoRaWAN HAT

d00m178, 2024-10-03:
hello!

is it possible to build 1ch GW on RPI4 Ubuntu 24.04 + Waveshare SX1262 LoRaWAN HAT and use it with Chirpstack Network Server for working with nodes based on Heltec Cubecell and RFM95 based nodes?

zhgzhg, 2024-10-04:
Greetings, @d00m178 !
If you mean [this HAT module](https://www.waveshare.com/sx1262-lorawan-hat.htm) - then yes, it's possible, but keep in mind this project doesn't implement reading from a GNSS, hence that part will remain unused.

d00m178, 2024-10-04:
@zhgzhg thank you for answer! 
yes, I have exactly this one Hat - LoRaWAN model.
I connected it to RPI4 + Ubuntu 24.04 but seems there is some issues and I dont know how to check if it really works or not.
I even dont understand how to check used SPI pins for prepare config file.
Please take a look:

1) I have Ubuntu 24.04 on RPI 4B 4Gb with Waveshare SX1262 LoRaWAN HAT connected (+ antennas connected to UFL connectors)
2) I installed raspi-config and enabled SPI:
1 - disabled
0 - enabled
```

### SPI NOT CONFIGURED:

# raspi-config nonint get_spi
1
# ls -la /dev/spi*
ls: cannot access '/dev/spi*': No such file or directory

# ls -la /sys/bus/spi/devices/
total 0

### ENABLE SPI:

# raspi-config nonint do_spi 0
# raspi-config nonint get_spi
0

# ls -la /dev/spi*
crw-rw---- 1 root dialout 153, 0 Oct  4 07:46 /dev/spidev0.0
crw-rw---- 1 root dialout 153, 1 Oct  4 07:46 /dev/spidev0.1

# ls -la /sys/bus/spi/devices/
total 0
drwxr-xr-x 2 root root 0 Oct  4 07:46 .
drwxr-xr-x 4 root root 0 Oct  4 07:27 ..
lrwxrwxrwx 1 root root 0 Oct  4 07:47 spi0.0 -> ../../../devices/platform/soc/fe204000.spi/spi_master/spi0/spi0.0
lrwxrwxrwx 1 root root 0 Oct  4 07:47 spi0.1 -> ../../../devices/platform/soc/fe204000.spi/spi_master/spi0/spi0.1

```
so I can see 2 spi devices, but not sure which is mine - which channel and port I need to use in config?

3) I installed wiringPi on 24.04 from repo - https://github.com/WiringPi/WiringPi

```
wget https://github.com/WiringPi/WiringPi/releases/download/3.10/wiringpi_3.10_arm64.deb
apt install ./wiringpi_3.10_arm64.deb
```
seems it works ok:

```
# gpio -v
gpio version: 3.10
Copyright (c) 2012-2024 Gordon Henderson and contributors
This is free software with ABSOLUTELY NO WARRANTY.
For details type: gpio -warranty

Hardware details:
  Type: Pi 4B, Revision: 05, Memory: 4096MB, Maker: Sony

System details:
  * Device tree present.
      Model: Raspberry Pi 4 Model B Rev 1.5
  * Supports full  user-level GPIO access via memory.
  * Supports basic user-level GPIO access via /dev/gpiomem.
  * Supports basic user-level GPIO access via /dev/gpiochip (slow).
```

```
# gpio readall
 +-----+-----+---------+------+---+---Pi 4B--+---+------+---------+-----+-----+
 | BCM | wPi |   Name  | Mode | V | Physical | V | Mode | Name    | wPi | BCM |
 +-----+-----+---------+------+---+----++----+---+------+---------+-----+-----+
 |     |     |    3.3v |      |   |  1 || 2  |   |      | 5v      |     |     |
 |   2 |   8 |   SDA.1 | ALT0 | 1 |  3 || 4  |   |      | 5v      |     |     |
 |   3 |   9 |   SCL.1 | ALT0 | 1 |  5 || 6  |   |      | 0v      |     |     |
 |   4 |   7 | GPIO. 7 |   IN | 1 |  7 || 8  | 1 | ALT5 | TxD     | 15  | 14  |
 |     |     |      0v |      |   |  9 || 10 | 1 | ALT5 | RxD     | 16  | 15  |
 |  17 |   0 | GPIO. 0 |   IN | 0 | 11 || 12 | 1 | IN   | GPIO. 1 | 1   | 18  |
 |  27 |   2 | GPIO. 2 |   IN | 0 | 13 || 14 |   |      | 0v      |     |     |
 |  22 |   3 | GPIO. 3 |   IN | 0 | 15 || 16 | 0 | IN   | GPIO. 4 | 4   | 23  |
 |     |     |    3.3v |      |   | 17 || 18 | 0 | IN   | GPIO. 5 | 5   | 24  |
 |  10 |  12 |    MOSI | ALT0 | 0 | 19 || 20 |   |      | 0v      |     |     |
 |   9 |  13 |    MISO | ALT0 | 1 | 21 || 22 | 0 | IN   | GPIO. 6 | 6   | 25  |
 |  11 |  14 |    SCLK | ALT0 | 0 | 23 || 24 | 1 | OUT  | CE0     | 10  | 8   |
 |     |     |      0v |      |   | 25 || 26 | 1 | OUT  | CE1     | 11  | 7   |
 |   0 |  30 |   SDA.0 |   IN | 1 | 27 || 28 | 1 | IN   | SCL.0   | 31  | 1   |
 |   5 |  21 | GPIO.21 |   IN | 1 | 29 || 30 |   |      | 0v      |     |     |
 |   6 |  22 | GPIO.22 |   IN | 1 | 31 || 32 | 0 | IN   | GPIO.26 | 26  | 12  |
 |  13 |  23 | GPIO.23 |   IN | 0 | 33 || 34 |   |      | 0v      |     |     |
 |  19 |  24 | GPIO.24 |   IN | 0 | 35 || 36 | 0 | IN   | GPIO.27 | 27  | 16  |
 |  26 |  25 | GPIO.25 |   IN | 0 | 37 || 38 | 0 | IN   | GPIO.28 | 28  | 20  |
 |     |     |      0v |      |   | 39 || 40 | 0 | IN   | GPIO.29 | 29  | 21  |
 +-----+-----+---------+------+---+----++----+---+------+---------+-----+-----+
 | BCM | wPi |   Name  | Mode | V | Physical | V | Mode | Name    | wPi | BCM |
 +-----+-----+---------+------+---+---Pi 4B--+---+------+---------+-----+-----+
```

4) then I clone your repo and just ran command "make" and it compiled application LoRaPktFwrd.

5) then I created config file from template - I just modified chip model and frequency because I dont know how to check SPI channel and port for my LoRaWAN Hat:

```
{
  "ic_model": "SX1262",

  "spi_speed_hz": 2000000,
  "spi_port": 1,
  "spi_channel": 0,

  "pin_nss_cs": 6,
  "pin_dio0": 9,
  "pin_dio1": 10,
  "pin_rest": -1,

  "spreading_factor": 7,
  "carrier_frequency_mhz": 868.0,
  "bandwidth_khz": 125.0,

  .........
  .........
```
 
4) and when I run LoRaPktFwrd it shows this:

```
# ./LoRaPktFwrd
(Fri Oct  4 07:21:00 2024) Started ./LoRaPktFwrd...

PlatformInfo:
------------

LoRa Packet Forwarder Version: next-b32a183

ID (EUI-64): d8:3a:dd:ff:ff:51:44:c8

Target LoRa Chip Model: SX1262

SPI Settings:
  SPI port=1
  SPI channel=0
  SPI clock speed=2000000 Hz

(WiringPI) Pins:
  nss_cs=6
  dio0=9
  dio1=10
  rest=-1

LoRa SX1262 Chip:
  Freq=868.000000 MHz
  BW=125.000 KHz
  SF=7
  CR=4/5
  SyncWord=0x34
  PreambleLength=8

Meta Information:
  Latitude=0.000000
  Longtitude=0.000000
  Altitude=10 meters
  Name/Definition: 1chan_uplink_pkt_fwd
  E-mail: contact@email.com
  Description: OPiPC LoRa 1-Ch GW

```

but on this LoRaWAN Hat there is switch "On-Off"
and the above command output appears in both switch positions.
so I wonder if it really works?

d00m178, 2024-10-04:
also I saw these examples in RadioLib repo (on which this project based on as far I understand):

for SX1278 which is mentioned in README of this project:
https://github.com/jgromes/RadioLib/blob/master/examples/SX127x/SX127x_PingPong/SX127x_PingPong.ino#L18C1-L23C40

```
// SX1278 has the following connections:
// NSS pin:   10
// DIO0 pin:  2
// NRST pin:  9
// DIO1 pin:  3
SX1278 radio = new Module(10, 2, 9, 3);
```
it also uses CS ( is it the same as NSS pin? ) pin and DIO0, DIO1 pins.

but for SX1262 it uses another pins:
https://github.com/jgromes/RadioLib/blob/master/examples/SX126x/SX126x_PingPong/SX126x_PingPong.ino#L18

```
// SX1262 has the following connections:
// NSS pin:   10
// DIO1 pin:  2
// NRST pin:  3
// BUSY pin:  9
SX1262 radio = new Module(10, 2, 3, 9);
```

and for example there no way to set BUSY pin in config.json for LoRaPktFwrd

also there is pinouts for my LoRaWAN HAT:
https://www.waveshare.com/wiki/SX1262_XXXM_LoRaWAN/GNSS_HAT#Pinout_Definition

d00m178, 2024-10-05:
ok I have some updates..
after spend a lot of time of searching, finally I edit config to this values:

```
# vi config.json
{
  "ic_model": "SX1262",

  "spi_speed_hz": 2000000,
  "spi_port": 0,
  "spi_channel": 0,

  "pin_nss_cs": 29,
  "pin_dio0": 22,
  "pin_dio1": 27,
  "pin_rest": 1,

  "spreading_factor": 7,
  "carrier_frequency_mhz": 868.0,
  "bandwidth_khz": 125.0,
  "coding_rate": 5,
  "sync_word": 52,
  "preamble_length": 8,
  .........
  .........
  .........
```

and after running LoRaPktFwrd I saw this:
```
# ./LoRaPktFwrd
(Sat Oct  5 15:40:54 2024) Started ./LoRaPktFwrd...

PlatformInfo:
------------

LoRa Packet Forwarder Version: next-b32a183

ID (EUI-64): d8:3a:dd:ff:ff:51:44:c8

Target LoRa Chip Model: SX1262

SPI Settings:
  SPI port=0
  SPI channel=0
  SPI clock speed=2000000 Hz

(WiringPI) Pins:
  nss_cs=29
  dio0=22
  dio1=27
  rest=1

LoRa SX1262 Chip:
  Freq=868.000000 MHz
  BW=125.000 KHz
  SF=7
  CR=4/5
  SyncWord=0x34
  PreambleLength=8

Meta Information:
  Latitude=0.000000
  Longtitude=0.000000
  Altitude=10 meters
  Name/Definition: 1chan_uplink_pkt_fwd
  E-mail: contact@email.com
  Description: OPiPC LoRa 1-Ch GW

LoRa chip setup succeeded! Waiting for data...

(Sat Oct  5 15:40:54 2024) Received UPlink packet:
 RSSI:                  -36.5 dBm
 SNR:                   8.500000 dB
 Frequency error:       0.000000 Hz
 Data:                  0 bytes

(Sat Oct  5 15:40:54 2024) Received UPlink packet:
 RSSI:                  -36.5 dBm
 SNR:                   8.500000 dB
 Frequency error:       0.000000 Hz
 Data:                  0 bytes

(Sat Oct  5 15:40:54 2024) Received UPlink packet:
 RSSI:                  -36.5 dBm
 SNR:                   8.500000 dB
 Frequency error:       0.000000 Hz
 Data:                  0 bytes

(Sat Oct  5 15:40:54 2024) Received UPlink packet:
 RSSI:                  -36.5 dBm
 SNR:                   8.500000 dB
 Frequency error:       0.000000 Hz
 Data:                  0 bytes

(Sat Oct  5 15:40:54 2024) Received UPlink packet:
 RSSI:                  -36.5 dBm
 SNR:                   8.500000 dB
 Frequency error:       0.000000 Hz
 Data:                  0 bytes

.............
............
.........

```

seems it runs ok but fisrt time it showed a lot of some strange UPLink packets errors.. idk what it means.
all next times when I run packet forwarder app - it shows only this:

```
# ./LoRaPktFwrd
(Sat Oct  5 16:20:09 2024) Started ./LoRaPktFwrd...

PlatformInfo:
------------

LoRa Packet Forwarder Version: next-b32a183

ID (EUI-64): d8:3a:dd:ff:ff:51:44:c8

Target LoRa Chip Model: SX1262

SPI Settings:
  SPI port=0
  SPI channel=0
  SPI clock speed=2000000 Hz

(WiringPI) Pins:
  nss_cs=29
  dio0=22
  dio1=27
  rest=1

LoRa SX1262 Chip:
  Freq=868.000000 MHz
  BW=125.000 KHz
  SF=7
  CR=4/5
  SyncWord=0x34
  PreambleLength=8

Meta Information:
  Latitude=0.000000
  Longtitude=0.000000
  Altitude=10 meters
  Name/Definition: 1chan_uplink_pkt_fwd
  E-mail: contact@email.com
  Description: OPiPC LoRa 1-Ch GW

LoRa chip setup succeeded! Waiting for data...
```

now I need to know - how to test it in LoRaWAN network? is it possible to add it to Chirpstack as GW?
also - is it possible to change frequency to 915 on this HAT?

d00m178, 2024-10-05:
it is strange.. sometimes after starting, LoRaPktFwrd shows output like this:
```

(Sat Oct  5 20:20:54 2024) No uplink ACK received from eu1.cloud.thethings.network
(Sat Oct  5 20:20:54 2024) Requeued the uplink packet.
(Sat Oct  5 20:20:56 2024) No uplink ACK received from eu1.cloud.thethings.network
(Sat Oct  5 20:20:56 2024) Requeued the uplink packet.
(Sat Oct  5 20:20:58 2024) No uplink ACK received from eu1.cloud.thethings.network
(Sat Oct  5 20:20:58 2024) Requeued the uplink packet.
(Sat Oct  5 20:21:00 2024) No uplink ACK received from eu1.cloud.thethings.network
(Sat Oct  5 20:21:00 2024) Requeued the uplink packet.
```

in the README of project there no example output how it should work in normal state..

also I checked Cubecell AB01 with Sender code:
https://github.com/HelTecAutomation/CubeCell-Arduino/blob/master/libraries/LoRa/examples/LoRaBasic/LoRaSender/LoRaSender.ino

but no luck - no activity in output of LoRaPktFwrd..
so maybe it doesnt work..

zhgzhg, 2024-10-06:
Hi @d00m178 , 
There are fair usage traffic policies within the things network and the LoRaWAN specification, which probably cause the server to temporarily ignore your uplinks, if they happen too often, which is most likely the reason why you are seeing those messages.

Also unfortunately, I have not added any LoRaWAN transmit examples, just plain LoRa. The latter can also cause problems with TTN.

However RadioLib makes it easier nowadays to work with LoRaWAN, because they include transmission examples. My recommendation is to use them as a basis. Compared to other popular alternatives like [Arduino LMIC](https://github.com/mcci-catena/arduino-lmic) , the RadioLib is significantly easier to customize. Customization is almost mandatory, because in order to achieve successful up/downlink communication, the trick is to pin your node to work only on the frequency and SF as the gateway works on. (a single channel frequency plan sort of speak)

d00m178, 2024-10-06:
@zhgzhg thank you for answer.

what about example of "normal" output of LoRaPktFwrd? 
how it should look like if everything is ok?
is it possible that LoRaPktFwrd can start with output that I provided above and dont work?

but for example - today it started and continues to shows these messages:

```
(Sun Oct  6 11:12:15 2024) Received UPlink packet:
 RSSI:                  -32.5 dBm
 SNR:                   24.500000 dB
 Frequency error:       0.000000 Hz
 Data:                  0 bytes

(Sun Oct  6 11:12:15 2024) Received UPlink packet:
 RSSI:                  -32.5 dBm
 SNR:                   24.500000 dB
 Frequency error:       0.000000 Hz
 Data:                  0 bytes

(Sun Oct  6 11:12:15 2024) Received UPlink packet:
 RSSI:                  -32.5 dBm
 SNR:                   24.500000 dB
 Frequency error:       0.000000 Hz
 Data:                  0 bytes

(Sun Oct  6 11:12:15 2024) Received UPlink packet:
 RSSI:                  -32.5 dBm
 SNR:                   24.500000 dB
 Frequency error:       0.000000 Hz
 Data:                  0 bytes

```
what it means?
looks like it receives some data from some device (but I have no transmitter powered on)
but when I check radio broadcast with my Nooelec RTL-SDR v5 - I can not see any signals in 868Mhz.

also how do you think - is my settings for SX1262 HAT correct or not?

```
  "ic_model": "SX1262",

  "spi_speed_hz": 2000000,
  "spi_port": 0,
  "spi_channel": 0,

  "_comment": "Waveshare SX1262 LoRaWAN HAT pinout:",
  "_comment": "https://www.waveshare.com/w/upload/5/56/SX1262_XXXM05.jpg",
  "_comment": "pin_nss_cs:      hw pin 40 - bcm 21 - wPi 29",
  "_comment": "pin_dio0 -> dio4: hw pin 31 - bcm 6 - wPi 22",
  "_comment": "pin_dio1:         hw pin 36 - bcm 16 - wPi 27",
  "_comment": "pin_rest(reset):  hw pin 12 - bcm 18 - wPi 1",

  "pin_nss_cs": 29,
  "pin_dio0": 22,
  "pin_dio1": 27,
  "pin_rest": 1,

  "spreading_factor": 7,
  "carrier_frequency_mhz": 868.0,
  "bandwidth_khz": 125.0,
  "coding_rate": 5,
  "sync_word": 52,
  "preamble_length": 8,
```

Also - you mentioned some "traffic policies" for TTN servers. I did not changed this part of configuration - so it means that if everything is ok with my setup - I can check the data on some web page on TTN? how to check it? I need to register to TTN and what next?

Also - I will check the RadioLib examples, but as I said I already checked LoRaSender example for my Heltec AB01 devices.
But LoRaPktFwrd doesnt hear anything and its output is empty tough SDR receiver shows activity from CubeCell AB01 on 868Mhz - so something wrong between this LoRaWAN HAT and CubeCell AB01.
Sure I configured frequency the same on both side.
I even changed “preamble length” to 12 and set “syncronize word” to 0x34 in the CubeCell examples of code…

So because of this I wonder how to check if this LoRaWAN HAT works with this application LoRaPktFwrd.

zhgzhg, 2024-10-07:
On the TTN topic - it seems you haven't gotten to that part yet. This project connects TTN / chirp stack / whatever LoRaWAN server to your LoRa nodes (e.g. cube cell). Those servers have web UI via which you can register and manage your gateways, nodes, and other apps which can consume or send IoT data in any direction. Via the web UI of the LoRaWAN server you can inspect the exchanged data too.
To accomplish above you need a registration there, as well as registering your gateway and any nodes that will communicate with it. Not doing so will cause the server to silently ignore any traffic sent to it.
Finally only after you have on-boarded your gateways and devices the traffic policies and the other LoRaWAN rules will apply for them (controlled by the LoRaWAN server).

---

Normally in the console logs for the packets you have received, the "Data:" output has to be greater than 0 bytes and when it is a hex dump will be printed right below the packet.

In your case the packets contain no data, which is not normal. Also, given the printed time, they seem to be coming in short bursts which is also not the case during a normal operation.

The RF settings (spreading factor, sync word, preamble length, and frequency) in the gateway must exactly match those set on the CubeCell.
But if you want to comply with the a standard LoRaWAN settings, the preamble and the sync word should be left to their default (template) values. And the transmit frequency should be one from the standard frequency plan [https://www.thethingsnetwork.org/docs/lorawan/frequency-plans/](https://www.thethingsnetwork.org/docs/lorawan/frequency-plans/).

zhgzhg, 2024-10-07:
@d00m178 , I have noticed in your config you are using DIO4 of the HAT as DIO0. This is likely the cause of the issue with empty data being received.

From the screenshots of the board it seems to be possible to bridge the real DIO0 to some of the unused GPIO pins of the raspberry hat, and then configure the pin in the config.json accordingly.

d00m178, 2024-10-07:
@zhgzhg than you very much for explaining)

as for DIO0 - do you mean that I need solder some pin from the chip to the pin on header?
it is no problem, but I need to know which exactly chip pin and which header pin.
also - on the pinout picture there no DIO0 pin:
https://www.waveshare.com/w/upload/5/56/SX1262_XXXM05.jpg

d00m178, 2024-10-07:
@zhgzhg 
this is what I receive with your settings:

```
# ./LoRaPktFwrd
(Mon Oct  7 18:57:06 2024) Started ./LoRaPktFwrd...

PlatformInfo:
------------

LoRa Packet Forwarder Version: next-b32a183

ID (EUI-64): d8:3a:dd:ff:ff:51:44:c8

Target LoRa Chip Model: SX1262

SPI Settings:
  SPI port=0
  SPI channel=0
  SPI clock speed=2000000 Hz

(WiringPI) Pins:
  nss_cs=29
  dio0=27
  dio1=22
  rest=1

LoRa SX1262 Chip:
  Freq=868.000000 MHz
  BW=125.000 KHz
  SF=7
  CR=4/5
  SyncWord=0x34
  PreambleLength=8

Meta Information:
  Latitude=0.000000
  Longtitude=0.000000
  Altitude=10 meters
  Name/Definition: 1chan_uplink_pkt_fwd
  E-mail: contact@email.com
  Description: OPiPC LoRa 1-Ch GW

LoRa chip setup failed, code (65534) Chip not found

```

but probably you a bit mistyped, it should be:

```
"pin_dio0": 22,
"pin_dio1": 27,
```

take a look on gpio output: 
```

# gpio readall
 +-----+-----+---------+------+---+---Pi 4B--+---+------+---------+-----+-----+
 | BCM | wPi |   Name  | Mode | V | Physical | V | Mode | Name    | wPi | BCM |
 +-----+-----+---------+------+---+----++----+---+------+---------+-----+-----+
 |     |     |    3.3v |      |   |  1 || 2  |   |      | 5v      |     |     |
 |   2 |   8 |   SDA.1 | ALT0 | 1 |  3 || 4  |   |      | 5v      |     |     |
 |   3 |   9 |   SCL.1 | ALT0 | 1 |  5 || 6  |   |      | 0v      |     |     |
 |   4 |   7 | GPIO. 7 |   IN | 1 |  7 || 8  | 1 | ALT5 | TxD     | 15  | 14  |
 |     |     |      0v |      |   |  9 || 10 | 1 | ALT5 | RxD     | 16  | 15  |
 |  17 |   0 | GPIO. 0 |   IN | 0 | 11 || 12 | 1 | OUT  | GPIO. 1 | 1   | 18  |
 |  27 |   2 | GPIO. 2 |   IN | 0 | 13 || 14 |   |      | 0v      |     |     |
 |  22 |   3 | GPIO. 3 |   IN | 0 | 15 || 16 | 0 | IN   | GPIO. 4 | 4   | 23  |
 |     |     |    3.3v |      |   | 17 || 18 | 0 | IN   | GPIO. 5 | 5   | 24  |
[... 169 lines cut ...]

  int state = radio.begin(868.0, 125.0, 7, 5, 0x34, 10, 8, 1.6, false);
```

and their radio-config the same like on my RPI+SX1262 HAT
and CubeCell's successfully communicates each other:

sender:
` [SX1262] Sending another packet ... transmission finished!`

receiver:
```
[SX1262] Received packet!
[SX1262] Data:		Hello World! #296
[SX1262] RSSI:		-58.00 dBm
[SX1262] SNR:		13.00 dB
[SX1262] Frequency error:	-23.25 Hz

```

and I suppose that LoRaPktFwrd also should print similar output like the CubeCell receiver but it is not.

> Perhaps there's still something wrong with with the wiring to the SX1262 HAT?

so this is a big question - how to configure this HAT with your gateway...

d00m178, 2024-10-07:
well.. this is strange..
I just decided to check this variant:

```
  "pin_nss_cs": 29,
  "pin_dio0": 27, --- HAT DIO1 pin
  "pin_dio1": 28, --- HAT BUSY pin
  "pin_rest": 1,

```
and it works!

[image]

d00m178, 2024-10-07:
upd: it is also strange - when the sender sends packets every seconds - gw receives every packet.
but if I configure send every 15 sec - gw could skip some number of packets..

zhgzhg, 2024-10-08:
Skipping some packets is actually a normal behavior for the gateway - it's a limitation of the software and hardware in that setup. (Even with a real multi-channel gateway this might happen.)
But it shouldn't be very noticeable once a node gets its LoRaWAN address, as typically the nodes are not allowed to transmit that often.

d00m178, 2024-10-08:
@zhgzhg - thank you for your help)
I will need to investigate this behavior because I need to send downlink message to my node and receive answer from it immediately.

btw - it is still interesting - why it start to work after changing pins - DIO1 instead DIO0 and BUSY instead DIO1.. and this is interesting also because you advised to do similar thing..

d00m178, 2024-10-08:
hello again @zhgzhg 

I have some new questions..
short variant - is this GW will work with TTN (for testing) and will it allow to register end nodes to TTN?

long variant:
As Im just started to exploring LoRaWAN I read a lot of different information about it.
Currently I learning it from RadioLib docs:
https://github.com/jgromes/RadioLib/blob/master/examples/LoRaWAN/LoRaWAN_Starter/notes.md
I prepared TTN account, created Application, created Device, created Gateway and prepared CubeCell AB01 with radiolib expamples:
https://github.com/jgromes/RadioLib/blob/master/examples/LoRaWAN/LoRaWAN_Starter/config.h
https://github.com/jgromes/RadioLib/blob/master/examples/LoRaWAN/LoRaWAN_Starter/LoRaWAN_Starter.ino

it started but cant join to TTN:
```
Copyright @2019-2020 Heltec Automation.All rights reserved.

Setup ... 
Initialise the radio
Join ('login') the LoRaWAN Network
Join failed - RADIOLIB_ERR_NO_JOIN_ACCEPT (-1116)
```

GW config:

```
# ./LoRaPktFwrd
(Tue Oct  8 21:23:26 2024) Started ./LoRaPktFwrd...

PlatformInfo:
------------

LoRa Packet Forwarder Version: next-b32a183

ID (EUI-64): d8:3a:dd:ff:ff:51:44:c8

Target LoRa Chip Model: SX1262

SPI Settings:
  SPI port=1
  SPI channel=0
  SPI clock speed=2000000 Hz

(WiringPI) Pins:
  nss_cs=29
  dio0=27
  dio1=28
  rest=1

LoRa SX1262 Chip:
  Freq=868.000000 MHz
  BW=125.000 KHz
  SF=7
  CR=4/5
  SyncWord=0x34
  PreambleLength=8

Meta Information:
  Latitude=0.000000
  Longtitude=0.000000
  Altitude=10 meters
  Name/Definition: 1chan_uplink_pkt_fwd
  E-mail: contact@email.com
  Description: OPiPC LoRa 1-Ch GW

LoRa chip setup succeeded! Waiting for data...

```
and I can see "Receive gateway status" messages in the Live data of Gateway in TTN console.
so it means that GW works and connected to TTN, right?

Also another CubeCell AB01 with another RadioLib firmware works and I can see messages from it in the GW output (my screenshots above)
https://github.com/jgromes/RadioLib/blob/master/examples/SX126x/SX126x_Transmit_Interrupt/SX126x_Transmit_Interrupt.ino

So GW works, radio between cubecell and GW also works.. but cubecell with LoRaWAN FW cant join to TTN..
Maybe Im something missed?

zhgzhg, 2024-10-09:
Hi @d00m178 ,
Shortly - yes.

The long version - first you got to adjust the RF frequencies to match your gateway's frequency plan. In your config 868.0 isn't a valid channel. Anything outsite the TTN / ChirpStack / etc frequency plans is likely to be ignored by the server.

Then you should try a simple ABP example [LoRaWAN_ABP.ino](https://github.com/jgromes/RadioLib/blob/master/examples/LoRaWAN/LoRaWAN_ABP/LoRaWAN_ABP.ino). Once you make it work and you can see the uplinks appearing in TTN's console you should try the LoRaWAN starter example.

I recommend that particular order because for a successfull OTAA the node has to be pinned to the RF parameters the single channel gateway listens on. Otherwise it will pick a channel randomly and the node may not to join the network from the first try. That pinning requires trickery with the frequency plan of the node...

d00m178, 2024-10-09:
hi @zhgzhg,

I have checked ABP example and it also doesnt work.
this is the settings of Device in TTN:

[image]

and this is configABP.h - all KEYs are from TTN Device settings:

[image]

and modem init in the main program:

`  int16_t state = radio.begin(868.1, 125.0, 9, 7, 0x34, 10, 8, 1.6, false);
`

and GW settings:

```
  "spreading_factor": 9,
  "carrier_frequency_mhz": 868.1,
  "bandwidth_khz": 125.0,
  "coding_rate": 7,
  "sync_word": 52,
  "preamble_length": 8,

LoRa SX1262 Chip:
  Freq=868.100000 MHz
  BW=125.000 KHz
  SF=9
  CR=4/7
  SyncWord=0x34
  PreambleLength=8
```

looks like everything is matching but CubeCell output is:

```
Sending uplink
No downlink received
Next uplink in 300 seconds
```
and no output on GW side and TTN shows - Never in the Last activity tab for Device..

zhgzhg, 2024-10-09:
If you don't see cli logs in from the GW, then the node likely transmits on another frequency. Those are defined within the RadioLib's frequency plan (RadioLib's frequency plan)[https://github.com/jgromes/RadioLib/blob/master/src/protocols/LoRaWAN/LoRaWANBands.cpp#L20].

Also the "coding_rate" is usually set to 5.

d00m178, 2024-10-09:
> node likely transmits on another frequency

I check radio broadcast with my Nooelec RTL-SDR v5 - and I can signals on 868.1Mhz from CubeCell Device ..
I decided to check CR == 7 because of this CR stated in the [default config](https://github.com/jgromes/RadioLib/wiki/Default-configuration#sx126x---lora-modem) for sx1262

but with CR == 5 it doesnt work in the same way - GW log empty while Device is trying to join.

Is it possible that my GW somehow works only with devices that work only with RAW LoRa?